jedie/inverter-connect #18: set-time command sets wrong register values

haraldfunk, 2023-04-30:
On my Deye Sun 600 inverter the set-time command sets wrong register values. 
After calling this command today (30.04.2023 14:20:22) output of the software looks like this:

┏━━━━┳━━━━━┳━━━━━┳━━━━┳━━━┓
┃ Counter ┃ Address ┃ Address ┃ Value ┃ Value ┃
┃               ┃  (hex)     ┃   (dec)    ┃ (hex)  ┃ (dec) ┃
┡━━━━━╇━━━━━╇━--━━━╇━-━━╇━━━━┩
│       1          │  0x16        │      22         │  06        │     6      │
│       2         │  0x17         │      23         │  17        │    23     │
│       3         │  0x18        │      24          │  04       │     4      │
│       4         │  0x19        │      25          │  1e        │    30     │
│       5         │  0x1a        │      26          │  0e        │    14     │
│       6         │  0x1b       │      27           │  14        │    20     │
└────┴-────┴─-─----─┴──-─┴ ───┘

First it seems that the addressing is wrongly shown as the values are written in 16 bit registers instead of 8 bit, so address 0x16 would contain 0x0617 and 0x17 contains 0x041e etc. after this call.

Second it is not clear to me where the 0x06 in register 0x16 is coming from. In my understanding register 0x16 should contain YY/MM, 0x17 should contain DD/HH and 0x18 should contain MM/SS. After the call  06/YY, MM/DD, HH/MM is written there instead. So seems like shifted for one byte by 0x06 in register 0x16. 

For me it should look like this:

━━━━┳━━━━━┳━━━━━┳━━━━┳━━━━┳━━━┓
┃ Counter ┃ Address ┃ Address ┃ Value ┃ Value ┃Value
┃               ┃  (hex)     ┃   (dec)    ┃ (hex)  ┃ (dec) ┃(parsed)
┡━━━━━╇━━━━━╇━--━━━╇━-━━╇━-━━╇━━━━┩
│       1          │  0x16        │      22         │  1704        │     5892      │ 23/04
│       2         │  0x17         │      23         │  1e0e        │    7694     │ 30/40
│       3         │  0x18        │      24          │  1416       │     5142      │ 20/22
└────┴-────┴─-─----─┴──-─ --┴ ──-─┴───┘

jedie, 2023-05-01:
Yes, the time set does not work properly.
In the forum also often people write that the registers are not updated.

But it looks like the day counter is reset anyway. That is actually the point.

I will build an automatism for this. That early in the morning always the time is set to reset the day counter.

But at the moment I try it via NTP.

It seems that you can specify an NTP server and that the time is really set correctly.

I added a example in the `print-at-commands` help:
```
    e.g.: Set NTP server, enable NTP and check the values:

    .../inverter-connect$ ./cli.py print-at-commands 192.168.123.456 NTPSER=192.168.1.1 NTPEN=on NTPSER NTPEN
```
in https://github.com/jedie/inverter-connect/commit/e64477b7e7a01dd0591786bd6bd4d19b768b9ba8

jedie, 2023-05-02:
Today i check the behavior:

A working NTP does not lead to a reset.
The `./cli.py set-time` works for me: The daily counter is set to 0.