jedie/inverter-connect #19: .\cli.py --help not working

AlfonsSchmid, 2023-05-01:
Hi I try to to run i with the raspberry pi. It always produces an error. Please see the output of the terminal. I updated to the newest version, but the result is the same. Do you have any idea?
Thanks and BR 
Alfons

pi@raspberrypi:~/inverter-connect $ python3.9 ./cli.py --help

+ /home/pi/inverter-connect/.venv/bin/pip-sync /home/pi/inverter-connect/requirements.dev.txt

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple
Collecting arrow==1.2.3 (from -r /tmp/tmprb_8bxwc (line 1))
  Using cached arrow-1.2.3-py3-none-any.whl (66 kB)
Collecting astor==0.8.1 (from -r /tmp/tmprb_8bxwc (line 4))
  Using cached https://www.piwheels.org/simple/astor/astor-0.8.1-py2.py3-none-any.whl (27 kB)
Collecting autopep8==2.0.2 (from -r /tmp/tmprb_8bxwc (line 7))
  Using cached autopep8-2.0.2-py2.py3-none-any.whl (45 kB)
Collecting binaryornot==0.4.4 (from -r /tmp/tmprb_8bxwc (line 10))
  Using cached https://www.piwheels.org/simple/binaryornot/binaryornot-0.4.4-py2.py3-none-any.whl (9.0 kB)
Collecting black==23.3.0 (from -r /tmp/tmprb_8bxwc (line 13))
  Using cached black-23.3.0-py3-none-any.whl (180 kB)
Collecting bleach==6.0.0 (from -r /tmp/tmprb_8bxwc (line 39))
  Using cached bleach-6.0.0-py3-none-any.whl (162 kB)
Collecting bx-py-utils==80 (from -r /tmp/tmprb_8bxwc (line 42))
  Using cached https://www.piwheels.org/simple/bx-py-utils/bx_py_utils-80-py3-none-any.whl (45 kB)
Collecting cachetools==5.3.0 (from -r /tmp/tmprb_8bxwc (line 45))
  Using cached cachetools-5.3.0-py3-none-any.whl (9.3 kB)
Collecting certifi==2022.12.7 (from -r /tmp/tmprb_8bxwc (line 48))
  Using cached certifi-2022.12.7-py3-none-any.whl (155 kB)
Collecting cffi==1.15.1 (from -r /tmp/tmprb_8bxwc (line 51))
  Using cached cffi-1.15.1.tar.gz (508 kB)
  Preparing metadata (setup.py) ... done
Collecting chardet==5.1.0 (from -r /tmp/tmprb_8bxwc (line 116))
  Using cached chardet-5.1.0-py3-none-any.whl (199 kB)
Collecting charset-normalizer==3.1.0 (from -r /tmp/tmprb_8bxwc (line 119))
  Using cached charset_normalizer-3.1.0-py3-none-any.whl (46 kB)
Collecting codespell==2.2.4 (from -r /tmp/tmprb_8bxwc (line 195))
  Using cached codespell-2.2.4-py3-none-any.whl (234 kB)
Collecting colorama==0.4.6 (from -r /tmp/tmprb_8bxwc (line 198))
  Using cached https://www.piwheels.org/simple/colorama/colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Collecting cookiecutter==2.1.1 (from -r /tmp/tmprb_8bxwc (line 201))
  Using cached cookiecutter-2.1.1-py2.py3-none-any.whl (36 kB)
Collecting coverage==7.2.3 (from -r /tmp/tmprb_8bxwc (line 204))
  Using cached coverage-7.2.3.tar.gz (757 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [66 lines of output]
      running dist_info
      creating /tmp/pip-modern-metadata-j03uhkpl/coverage.egg-info
      writing /tmp/pip-modern-metadata-j03uhkpl/coverage.egg-info/PKG-INFO
      writing dependency_links to /tmp/pip-modern-metadata-j03uhkpl/coverage.egg-info/dependency_links.txt
      writing entry points to /tmp/pip-modern-metadata-j03uhkpl/coverage.egg-info/entry_points.txt
      writing requirements to /tmp/pip-modern-metadata-j03uhkpl/coverage.egg-info/requires.txt
      writing top-level names to /tmp/pip-modern-metadata-j03uhkpl/coverage.egg-info/top_level.txt
      writing manifest file '/tmp/pip-modern-metadata-j03uhkpl/coverage.egg-info/SOURCES.txt'
      reading manifest file '/tmp/pip-modern-metadata-j03uhkpl/coverage.egg-info/SOURCES.txt'
      reading manifest template 'MANIFEST.in'
      no previously-included directories found matching 'doc/_build'
      no previously-included directories found matching 'doc/_spell'
      no previously-included directories found matching 'tests/eggsrc/build'
      adding license file 'LICENSE.txt'
      writing manifest file '/tmp/pip-modern-metadata-j03uhkpl/coverage.egg-info/SOURCES.txt'
      creating '/tmp/pip-modern-metadata-j03uhkpl/coverage-7.2.3.dist-info'
      Traceback (most recent call last):
        File "/home/pi/inverter-connect/.venv/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
[... 60 lines cut ...]
  File "/home/pi/inverter-connect/.venv/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/inverter-connect/.venv/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/pi/inverter-connect/.venv/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/inverter-connect/.venv/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/pi/inverter-connect/.venv/lib/python3.10/site-packages/piptools/scripts/sync.py", line 174, in cli
    sync.sync(
  File "/home/pi/inverter-connect/.venv/lib/python3.10/site-packages/piptools/sync.py", line 244, in sync
    run(  # nosec
  File "/usr/local/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/home/pi/inverter-connect/.venv/bin/python', '-m', 'pip', 'install', '-r', '/tmp/tmprb_8bxwc', '--extra-index-url', 'https://www.piwheels.org/simple']' returned non-zero exit status 1.
Traceback (most recent call last):
  File "/home/pi/inverter-connect/./cli.py", line 115, in <module>
    main(sys.argv)
  File "/home/pi/inverter-connect/./cli.py", line 101, in main
    verbose_check_call(PIP_SYNC_PATH, str(DEP_LOCK_PATH))
  File "/home/pi/inverter-connect/./cli.py", line 81, in verbose_check_call
    return subprocess.check_call(popen_args)
  File "/usr/local/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '(PosixPath('/home/pi/inverter-connect/.venv/bin/pip-sync'), '/home/pi/inverter-connect/requirements.dev.txt')' returned non-zero exit status 1.

jedie, 2023-05-02:
Sorry, i have no idea. A little bit strange looks this lines:
[image]

And there is a Python version mismatch: You start the cli with **python3.9**, but later in the traceback you can see that **python3.10** is used.

Maybe try 3.10, e.g.:
```
pi@raspberrypi:~/inverter-connect $ rm -Rf .venv
pi@raspberrypi:~/inverter-connect $ python3.10 ./cli.py --help
```
What's the output of: `python3 --version` ? On my Pi it's 3.9.2, but i used https://dietpi.com/

jedie, 2023-05-03:
Hm. Rust toolchain is neede, because [requirements.dev.txt](https://github.com/jedie/inverter-connect/blob/main/requirements.dev.txt) is installed. We have to split the normal use case. So i made #22 for this.

Your output suggest that "cryptography" needs Rust. But on my Pi i haven't Rust installed.

Think the trick is to enable https://www.piwheels.org/
Just create the file `/etc/pip.conf` with the content:
```
[global]
extra-index-url=https://www.piwheels.org/simple
```

Think i should add this to README

giovanne123, 2023-05-03:
FYI, on my Pi already exists.
[image]

Btw: in addition I don't want to modify python directly on my OS-level to prevent other stuff stop working ;-)

jedie, 2023-05-03:
> FYI, on my Pi already exists.

Hm. Sad.
Than i don't know what's the difference between plain Raspian and my dietpi installation :(

> Btw: in addition I don't want to modify python directly on my OS-level to prevent other stuff stop working ;-)

That's why the script creates and used a python virtualenv ;)

jedie, 2023-05-03:
@AlfonsSchmid @giovanne123 I started to tackle #22 in https://github.com/jedie/inverter-connect/pull/23

You can already tried it on your Pi by switch to the "fix#22" branch, e.g.:
```
~/inverter-connect$ git fetch
~/inverter-connect$ git checkout origin/fix#22
~/inverter-connect$ ./cli.py --help
```

AlfonsSchmid, 2023-05-03:
@jedie, thanks a lot. I will try and give a feedback. Please give me a little time.

giovanne123, 2023-05-03:
cli.py - is working for me now
(dev-cli.py - still isn't working, will have a look for it later...)

jedie, 2023-05-03:
dev-cli.py ist only needed for developers 😉

closed 2023-05-04 by jedie