jedie/inverter-connect #22: Split between "normal" and "dev" mode

jedie, 2023-05-03:
So, the normal mode will only install [requirements.txt](https://github.com/jedie/inverter-connect/blob/main/requirements.txt)
and only the dev mode: [requirements.dev.txt](https://github.com/jedie/inverter-connect/blob/main/requirements.dev.txt)

that will help to fix https://github.com/jedie/inverter-connect/issues/19

giovanne123, 2023-05-03:
FYI, started with requirements.txt edited in cli.py without dev:

```
pi@raspberrypi:/home/pi/inverter-connect# python3.9 ./cli.py --help

+ /home/pi/inverter-connect/.venv/bin/pip-sync /home/pi/inverter-connect/requirements.txt

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple
Collecting bx-py-utils==80 (from -r /tmp/tmpffsm6p3p (line 1))
  Using cached https://www.piwheels.org/simple/bx-py-utils/bx_py_utils-80-py3-none-any.whl (45 kB)
Collecting markdown-it-py==2.2.0 (from -r /tmp/tmpffsm6p3p (line 4))
  Using cached markdown_it_py-2.2.0-py3-none-any.whl (84 kB)
Collecting mdurl==0.1.2 (from -r /tmp/tmpffsm6p3p (line 7))
  Using cached https://www.piwheels.org/simple/mdurl/mdurl-0.1.2-py3-none-any.whl (10.0 kB)
Collecting paho-mqtt==1.6.1 (from -r /tmp/tmpffsm6p3p (line 10))
  Using cached paho_mqtt-1.6.1-py3-none-any.whl
Collecting pygments==2.15.1 (from -r /tmp/tmpffsm6p3p (line 12))
  Using cached Pygments-2.15.1-py3-none-any.whl (1.1 MB)
Collecting pyyaml==6.0 (from -r /tmp/tmpffsm6p3p (line 15))
  Using cached PyYAML-6.0-cp39-cp39-linux_armv7l.whl
Collecting rich==13.3.5 (from -r /tmp/tmpffsm6p3p (line 56))
  Using cached rich-13.3.5-py3-none-any.whl (238 kB)
Collecting rich-click==1.6.1 (from -r /tmp/tmpffsm6p3p (line 59))
  Using cached rich_click-1.6.1-py3-none-any.whl (19 kB)
Requirement already satisfied: click>=7 in ./.venv/lib/python3.9/site-packages (from rich-click==1.6.1->-r /tmp/tmpffsm6p3p (line 59)) (8.1.3)
Installing collected packages: paho-mqtt, pyyaml, pygments, mdurl, bx-py-utils, markdown-it-py, rich, rich-click
Successfully installed bx-py-utils-80 markdown-it-py-2.2.0 mdurl-0.1.2 paho-mqtt-1.6.1 pygments-2.15.1 pyyaml-6.0 rich-13.3.5 rich-click-1.6.1

+ /home/pi/inverter-connect/.venv/bin/pip install --no-deps -e .

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi/inverter-connect
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Installing backend dependencies ... done
  Preparing editable metadata (pyproject.toml) ... done
Building wheels for collected packages: inverter-connect
  Building editable for inverter-connect (pyproject.toml) ... done
  Created wheel for inverter-connect: filename=inverter_connect-0.3.1-0.editable-py3-none-any.whl size=5251 sha256=6afe493615e198f6b16431a187d29ef03a1d2e82f17a16090abfe52d9929b5fc
  Stored in directory: /tmp/pip-ephem-wheel-cache-9kjguiku/wheels/cf/6c/71/c67673286534843b8ac39160e5cb4e788315f29832e1ad3bf8
Successfully built inverter-connect
Installing collected packages: inverter-connect
Successfully installed inverter-connect-0.3.1

+ /home/pi/inverter-connect/.venv/bin/inverter --help

Traceback (most recent call last):
  File "/home/pi/inverter-connect/.venv/bin/inverter", line 5, in <module>
    from inverter.__main__ import main
  File "/home/pi/inverter-connect/inverter/__main__.py", line 7, in <module>
    from inverter.cli import cli_app
  File "/home/pi/inverter-connect/inverter/cli/cli_app.py", line 10, in <module>
    from manageprojects.utilities import code_style
ModuleNotFoundError: No module named 'manageprojects'

```

jedie, 2023-05-03:
Yes. It not only the non-dev-requirements ... The CLI must be also split and all dev-commands should be added only in dev mode.

jedie, 2023-05-03:
Seems that it's possible just by use two "cli" scripts: https://github.com/jedie/inverter-connect/pull/23

But the tests and the README needs some updates ;)

jedie, 2023-05-03:
done with https://github.com/jedie/inverter-connect/pull/23

closed 2023-05-03 by jedie