jedie/inverter-connect #26: UnboundLocalError

CarstenGrohmann, 2023-05-05:
Hi,

I got this error message, as I tried [some additional entries](https://github.com/StephanJoubert/home_assistant_solarman/pull/285) for deye 2mppt.

```
# ./cli.py print-values inverter 

+ /home/carsten/inverter-connect/.venv-app/bin/inverter_app print-values inverter

inverter v0.4.0
INFO:inverter.connection:Connect to inverter:48899...
InverterInfo(ip='192.168.xxx.xxx', mac='xxxxxxxxxxxx', serial=xxxxxxxxxx)

        * PV1 Voltage                    : 33.0 V      (Register: 006D, length: 1)
        * PV2 Voltage                    : 0.0 V       (Register: 006F, length: 1)
        * PV1 Current                    : 0.8 A       (Register: 006E, length: 1)
        * PV2 Current                    : 0.0 A       (Register: 0070, length: 1)
        * Daily Production               : 30.7 kWh    (Register: 003C, length: 1)
        * Daily Production 1             : 30.7 kWh    (Register: 0041, length: 1)
        * Daily Production 2             : 0.0 kWh     (Register: 0042, length: 1)
        * Total Production               : 35.9 kWh    (Register: 003F, length: 2)
WARNING:inverter.api:Parser error with response=ModbusResponse(slave_id=1, modbus_function=3, data_hex='0167') parameter=Parameter(start_register=69, length=1, group='solar', name='Total Production 1', device_class='energy', state_class='total_increasing', unit='kWh', scale=0.1, parser=<function parse_swapped_number at 0x7f83f168fbe0>, offset=None, lookup=None): Wrong len 4: data_hex='0167' - retry...
WARNING:inverter.api:Parser error with response=ModbusResponse(slave_id=1, modbus_function=3, data_hex='0167') parameter=Parameter(start_register=69, length=1, group='solar', name='Total Production 1', device_class='energy', state_class='total_increasing', unit='kWh', scale=0.1, parser=<function parse_swapped_number at 0x7f83f168fbe0>, offset=None, lookup=None): Wrong len 4: data_hex='0167' - retry...
WARNING:inverter.api:Parser error with response=ModbusResponse(slave_id=1, modbus_function=3, data_hex='0167') parameter=Parameter(start_register=69, length=1, group='solar', name='Total Production 1', device_class='energy', state_class='total_increasing', unit='kWh', scale=0.1, parser=<function parse_swapped_number at 0x7f83f168fbe0>, offset=None, lookup=None): Wrong len 4: data_hex='0167' - retry...

Signing off with "AT+Q"...Goodbye ;)

Traceback (most recent call last):
  File "/home/carsten/inverter-connect/.venv-app/bin/inverter_app", line 8, in <module>
    sys.exit(main())
  File "/home/carsten/inverter-connect/inverter/__main__.py", line 11, in main
    cli_app.main()
  File "/home/carsten/inverter-connect/inverter/cli/cli_app.py", line 422, in main
    cli()
  File "/home/carsten/inverter-connect/.venv-app/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/carsten/inverter-connect/.venv-app/lib/python3.10/site-packages/rich_click/rich_group.py", line 21, in main
    rv = super().main(*args, standalone_mode=False, **kwargs)
  File "/home/carsten/inverter-connect/.venv-app/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/carsten/inverter-connect/.venv-app/lib/python3.10/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/carsten/inverter-connect/.venv-app/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/carsten/inverter-connect/.venv-app/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/carsten/inverter-connect/inverter/cli/cli_app.py", line 102, in print_values
    for value in inverter:
  File "/home/carsten/inverter-connect/inverter/api.py", line 110, in __iter__
    raise Exception from err  # noqa
UnboundLocalError: local variable 'err' referenced before assignment
```

Please solve this issue,

Thank you,
Carsten

BTW: The definition file is "wrong" as I have a deye 1mppt inverter. I'll strip down it later and file a PR for home_assistant_solarman.

jedie, 2023-05-20:
Think, this is fixed in the past ;)

closed 2023-05-20 by jedie

CarstenGrohmann, 2023-05-23:
Thank you for all these changes!