jedie/inverter-connect #31: print-at-commands shows wrong results after the first AT command failed

CarstenGrohmann, 2023-05-06:
Hi,

Today the first query of an AT command failed. As a result, the wrong values were displayed for the subsequent AT commands. For the current command the result of the previously executed command was displayed - as this example shows:

```
# ./cli.py print-at-commands inverter 

+ /home/carsten/inverter-connect/.venv-app/bin/inverter_app print-at-commands inverter

inverter v0.4.0
INFO:inverter.connection:Connect to inverter:48899...
InverterInfo(ip='192.168.xxx.xxx', mac='xxxxxxxxxxxx', serial=xxxxxxxxxx)

        * AT+VER       : WARNING:inverter.connection:Get no response from inverter: timed out - retry...
retry...-1
        * AT+BVER      : HF4.13.23-YZ (2023-03-01 15:00 2M)
        * AT+HWVER     : Non-standard version, 0xC1272
        * AT+WEBVER    : U06,6
        * AT+YZVER     : V1.0.24
        * AT+PING      : MW3_16U_5406_1.56
        * AT+TIME      : -3
        * AT+NDBGS     : 5,60,120
        * AT+WIFI      : 0
        * AT+WMODE     : UP
        * AT+WEBU      : STA
[...]
```

Expected result:

```
# ./cli.py print-at-commands inverter 

+ /home/carsten/inverter-connect/.venv-app/bin/inverter_app print-at-commands inverter

inverter v0.4.0
INFO:inverter.connection:Connect to inverter:48899...
InverterInfo(ip='192.168.xxx.xxx', mac='xxxxxxxxxxxx', serial=xxxxxxxxxx)

        * AT+VER       : WARNING:inverter.connection:Get no response from inverter: timed out - retry...
retry...-1
        * AT+BVER      : Non-standard version, 0xC1272
        * AT+HWVER     : U06,6
        * AT+WEBVER    : V1.0.24
        * AT+YZVER     : MW3_16U_5406_1.56
        * AT+PING      : -3
        * AT+TIME      : 5,60,120
        * AT+NDBGS     : 0
        * AT+WIFI      : UP
        * AT+WMODE     : STA
[...]
```

I use a Deye SUN300G3 EU 230 inverter.

That do you think about this behaviour?

Regards,
Carsten

jedie, 2023-05-07:
Strange. Think i have to refactor the retry implementation here...

jedie, 2023-05-20:
this is fixed

closed 2023-05-20 by jedie