jedie/inverter-connect #48: Calling edit-settings fails if no config file exists

CarstenGrohmann, 2023-05-19:
Configuration handling fails, if the configuration file does not exists.

```
# ./cli.py --help

+ /home/carsten/inverter-connect/.venv-app/bin/inverter_app --help

Use user settings file: /home/carsten/.config/inverter-connect/inverter-connect.toml...

╭─────────────────────────────────────────── Error ────────────────────────────────────────────╮
│                                                                                              │
│                                                                                              │
│     [yellow]No settings created yet[/yellow]: [green](Hint: call "edit-settings" first!)     │
│                                                                                              │
│                                                                                              │
╰──────────────────────────────────────────────────────────────────────────────────────────────╯

# ./cli.py edit-settings

+ /home/carsten/inverter-connect/.venv-app/bin/inverter_app edit-settings

Use user settings file: /home/carsten/.config/inverter-connect/inverter-connect.toml...

╭─────────────────────────────────────────── Error ────────────────────────────────────────────╮
│                                                                                              │
│                                                                                              │
│     [yellow]No settings created yet[/yellow]: [green](Hint: call "edit-settings" first!)     │
│                                                                                              │
│                                                                                              │
╰──────────────────────────────────────────────────────────────────────────────────────────────╯

# ll ~/.config/inverter-connect/
insgesamt 0
```

Affected version:
* current main branch

Workaround:
* create empty configuration file
 
    ```
    # touch ~/.config/inverter-connect/inverter-connect.toml
    ```

* execute `./cli.py edit-settings` again
 

What do you think about this behavior?

Regards,
Carsten

mgraupner, 2023-05-20:
I'm having the exactly same problem. Your solution solved it.

If it helps: Running MacOS and Python 3.9.

jedie, 2023-05-20:
Hm. Okay i will try to reproduce it. Thanks for reporting.

jedie, 2023-05-20:
fixed by https://github.com/jedie/inverter-connect/pull/52

closed 2023-05-20 by jedie