jedie/inverter-connect #60: Questions for pr #27 'Clear "Daily Production" every night by set time'

CarstenGrohmann, 2023-05-25:
Hi @jedie,

I'm not sure if I understand the code right. Every time (in every iteration of the while loop in `publish_forever()`) you enter the context of DailyProductionReset, and if it's between 1am and 3am, you'll reset the "Daily Production" counter. 

Would it be a better solution to track the date of the last reset and check in `DailyProductionReset.__enter__()` if you did a reset at the same day already? This may reduce the number of resets at night.

I use a smaller inverter a SUN300 and this is offline during night. What's your suggested approch for such a small inverter to reset the "Daily Production" counter?

Regards,
Carsten

jedie, 2023-05-26:
Yes, a good idea!
Also because i have the SUN600 and he is also offline at night...

jedie, 2023-05-26:
One Problem: What's if the status is lost? e.g.: The computer was rebooted or the services restarts of some other reason.

So you can't just init with "no reset made today" and toggle this boolean flag. Because every restart will reset the daily production, too.

So we have to store the state on disk to avoid this. Maybe just the date of the last reset into e.g.: `~/.config/inverter-connect/lastreset`

jedie, 2023-05-26:
I implement the idea in https://github.com/jedie/inverter-connect/pull/61

We will see tomorrow if it's work ;)

jedie, 2023-05-27:
Hm. Doesn't work, today:

[image]

But the reset was made, today:

```
cat ~/.config/inverter-connect/daily_reset_state.txt 
2023-05-27
```

Hm...

CarstenGrohmann, 2023-05-28:
The original firmware resets the "daily production" at the first connection with the central server. With this implementation, you lose production between the first sun and the time you connect home. Yesterday, I re-enabled communication with the internet for the inverter, and the inverter resets the day's production immediately after. 

What do you think about a similar approach by running "Set-Time" once a day, just after the software was able to connect to the inverter for the first time?

Using the status file to ensure that "Set-Time" is only run once a day, even if the software has been restarted.

The result would be very similar to the original firmware when the inverter is connected to the internet. There will always be a small amount not reported. If the software runs continuously, the amount is very small probably under 5 minutes of harvested energy at the beginning of each day.

jedie, 2023-05-30:
> What do you think about a similar approach by running "Set-Time" once a day, just after the software was able to connect to the inverter for the first time?
> Using the status file to ensure that "Set-Time" is only run once a day, even if the software has been restarted.

This is exactly how it's implemented, now ;) But doesn't work. Don't know why.

commit 1e2d8b19 (2023-05-26): Implement a new "Daily reset" solution

Save the information that the reset was done into a file and try the reset independ of the time.
Just compare the current date.

So a reset will be made every day as long as it was successfully, then it's skipped until the next
day.

Closes #60
PR #61 (2023-05-26): Implement a new "Daily reset" solution
Save the information that the reset was done into a file and try the reset independ of the time. Just compare the current date.

So a reset will be made every day as long as it was successfully, then it's skipped until the next day.

Closes #60