jedie/inverter-connect #73: Automatic installation with Ansible fails

CarstenGrohmann, 2023-07-24:
Hi Jens,

currently I try to automate the installation of inverter-connect with Ansible. This causes this error:
```
fatal: [rpi4]: FAILED! => changed=true 
  cmd:
  - ./cli.py
  - print-values
  delta: '0:03:27.252071'
  end: '2023-07-24 15:17:17.989186'
  msg: non-zero return code
  rc: 1
  start: '2023-07-24 15:13:50.737115'
  stderr: |-
    Traceback (most recent call last):
      File "/home/spower/inverter-connect/.venv-app/bin/inverter_app", line 5, in <module>
        from inverter.__main__ import main
      File "/home/spower/inverter-connect/inverter/__main__.py", line 7, in <module>
        from inverter.cli import cli_app
      File "/home/spower/inverter-connect/inverter/cli/cli_app.py", line 96, in <module>
        toml_settings = TomlSettings(
      File "/home/spower/inverter-connect/.venv-app/lib/python3.9/site-packages/ha_services/toml_settings/api.py", line 40, in __init__
        self.file_path = self.get_settings_file_path(dir_name, file_name)
      File "/home/spower/inverter-connect/.venv-app/lib/python3.9/site-packages/ha_services/toml_settings/api.py", line 60, in get_settings_file_path
        settings_path.mkdir(parents=False, exist_ok=False)
      File "/usr/lib/python3.9/pathlib.py", line 1312, in mkdir
        self._accessor.mkdir(self, mode)
    PermissionError: [Errno 13] Permission denied: '/home/ansible/inverter-connect'
```

My installation steps are:

1. As user ansible: Create a user "spower"
2. As user spower: 
    1. Check out repo to `~spower/inverter-connect/`
    2. Copy configuration file to `~spower/.config/inverter-connect/inverter-connect.toml`
    3. Run `cli.py print-values` to create virtual environment and to test the connectivity to the inverter
    4. Permission error occurs, as shown in the error message

The Ansible artifact is:
```
    - name: Install and configure jedie/inverter
      become: true
      become_user: "{{ solar_user_name }}"
      vars:
        mqtt_server: localhost
        solar_user_name: spower
        solar_user_uid: 2123
        solar_group_name: spower
        solar_group_gid: 2123
        inverter_path: "~spower/inverter-connect/"
      block:
        - name: Clone jedie/inverter repository
          ansible.builtin.git: # noqa: latest[git]
            repo: https://github.com/jedie/inverter-connect/
            dest: "{{ inverter_path }}"
        - name: Create configuration directory ~{{ solar_user_name }}/.config/inverter-connect/
          ansible.builtin.file:
            path: "~{{ solar_user_name }}/.config/inverter-connect/"
            state: directory
            mode: '0755'
        - name: Apply configuration for jedie/inverter
          ansible.builtin.template:
            src: inverter-connect.toml
            dest: "~{{ solar_user_name }}/.config/inverter-connect/inverter-connect.toml"
            owner: "{{ solar_user_name }}"
            group: "{{ solar_group_name }}"
            mode: "0640"
        - name: Test connectivity to inverter
          ansible.builtin.command:
            cmd: "./cli.py print-values"
          register: _output
          changed_when: _output.rc != 0
          args:
            chdir: "{{ inverter_path }}"
          environment:
            NO_COLOR: 1
```

The error is caused by the special handling of `~` in combination with environment variable `SUDO_USER` as coded here:
https://github.com/jedie/ha-services/blob/4ad2a1b56d7b521e80dbe4ec3ecebac0440ac2df/ha_services/cli_tools/path_utils.py#L35-L54

From my perspective I access the system with the user "ansible" and use sudo to "spower". It's not intended to store anything in the home directory of the ansible user.

That do you think about this issue?

Regards,
Carsten

CarstenGrohmann, 2023-07-27:
My current workaround is:

```
- name: Test connectivity to inverter
  ansible.builtin.shell:
    cmd: 'su - {{ solar_user_name }} -c "cd {{ inverter_path }} && ./cli.py print-values"
  become_user: root
  register: _output
  changed_when: _output.rc != 0
  args:
    chdir: "{{ inverter_path }}"
  environment:
    NO_COLOR: "1"
```

jedie, 2023-08-08:
I have no Ansible knowledge. So i can't help here. Will move this to Discussion.