jedie/inverter-connect #75: FileNotFoundError for service_template.txt

CarstenGrohmann, 2023-07-25:
Hi Jens,

I got this error:

```
$ ./cli.py systemd-debug 

+ /home/spower/inverter-connect/.venv-app/bin/inverter_app systemd-debug

Use user settings file: /home/spower/.config/inverter-connect/inverter-connect.toml.read, ok.
inverter v0.11.1
                                                            (Set log level 0: ERROR)

╭─────────────────────── Traceback (most recent call last) ────────────────────────╮
│ /home/spower/inverter-connect/.venv-app/bin/inverter_app:8 in <module>           │
│                                                                                  │
│   5 from inverter.__main__ import main                                           │
│   6 if __name__ == '__main__':                                                   │
│   7 │   sys.argv[0] = re.sub(r'(-script\.pyw|\.exe)?$', '', sys.argv[0])         │
│ ❱ 8 │   sys.exit(main())                                                         │
│   9                                                                              │
│                                                                                  │
│ ╭─────────────────────── locals ───────────────────────╮                         │
│ │ main = <function main at 0x7fba053310>               │                         │
│ │   re = <module 're' from '/usr/lib/python3.9/re.py'> │                         │
│ │  sys = <module 'sys' (built-in)>                     │                         │
│ ╰──────────────────────────────────────────────────────╯                         │
│                                                                                  │
│ /home/spower/inverter-connect/inverter/__main__.py:11 in main                    │
│                                                                                  │
│    8                                                                             │
│    9                                                                             │
│   10 def main():                                                                 │
│ ❱ 11 │   cli_app.main()                                                          │
│   12                                                                             │
│   13                                                                             │
│   14 if __name__ == '__main__':                                                  │
│                                                                                  │
│                             ... 11 frames hidden ...                             │
│                                                                                  │
│ /usr/lib/python3.9/pathlib.py:1241 in open                                       │
│                                                                                  │
│   1238 │   │   Open the file pointed by this path and return a file object, as   │
│   1239 │   │   the built-in open() function does.                                │
│   1240 │   │   """                                                               │
│ ❱ 1241 │   │   return io.open(self, mode, buffering, encoding, errors, newline,  │
│   1242 │   │   │   │   │      opener=self._opener)                               │
│   1243 │                                                                         │
│   1244 │   def read_bytes(self):                                                 │
│                                                                                  │
│ ╭─────────────────────────────────── locals ───────────────────────────────────╮ │
│ │ buffering = -1                                                               │ │
│ │  encoding = 'UTF-8'                                                          │ │
│ │    errors = None                                                             │ │
│ │      mode = 'r'                                                              │ │
│ │   newline = None                                                             │ │
│ │      self = PosixPath('~spower/inverter-connect/.venv-app/lib/python3.9/sit… │ │
│ ╰──────────────────────────────────────────────────────────────────────────────╯ │
│                                                                                  │
│ /usr/lib/python3.9/pathlib.py:1109 in _opener                                    │
│                                                                                  │
│   1106 │                                                                         │
│   1107 │   def _opener(self, name, flags, mode=0o666):                           │
│   1108 │   │   # A stub for the opener argument to built-in open()               │
│ ❱ 1109 │   │   return self._accessor.open(self, flags, mode)                     │
│   1110 │                                                                         │
│   1111 │   def _raw_open(self, flags, mode=0o777):                               │
│   1112 │   │   """                                                               │
│                                                                                  │
│ ╭─────────────────────────────────── locals ───────────────────────────────────╮ │
│ │ flags = 524288                                                               │ │
│ │  mode = 438                                                                  │ │
│ │  name = '~spower/inverter-connect/.venv-app/lib/python3.9/site-packages/ha_… │ │
│ │  self = PosixPath('~spower/inverter-connect/.venv-app/lib/python3.9/site-pa… │ │
│ ╰──────────────────────────────────────────────────────────────────────────────╯ │
╰──────────────────────────────────────────────────────────────────────────────────╯
FileNotFoundError: [Errno 2] No such file or directory: 
'~spower/inverter-connect/.venv-app/lib/python3.9/site-packages/ha_services/systemd/
service_template.txt'
──────────────────────────── Tue 25 Jul 2023 08:56:05  ─────────────────────────────
```

as I've specified the template path (and other paths too) in the configuration file like:

```
$ grep template ~/.config/inverter-connect/inverter-connect.toml 
template_path = "~spower/inverter-connect//.venv-app/lib/python3.9/site-packages/ha_services/systemd/service_template.txt"
[systemd.template_context]
```

The file itself exists:
```
$ ll ~spower/inverter-connect/.venv-app/lib/python3.9/site-packages/ha_services/systemd/service_template.txt
-rw-r--r-- 1 spower spower 320 Jul 24 15:15 /home/spower/inverter-connect/.venv-app/lib/python3.9/site-packages/ha_services/systemd/service_template.txt
```

Please add support for expanding user paths like `~spower`.

Thanks,
Carsten

jedie, 2023-08-08:
This is implemented in upstream project: https://github.com/jedie/ha-services/releases/tag/v0.3.3

jedie, 2023-08-08:
It's not supported and released as v0.12.0

commit c2c7419a (2023-08-08): Update requirements

ha-services>=0.3.3 will also fix #75
PR #87 (2023-08-08): Update requirements
ha-services>=0.3.3 will also fix #75
closed 2023-08-08 by jedie