jedie/inverter-connect #119: pip fails installing dependencies

prodigy, 2024-02-11:
**Python version:**
```
(venv) PS E:\Tools\Solar\inverter-connect> python --version
Python 3.10.11
(venv) PS E:\Tools\Solar\inverter-connect> pip --version
pip 24.0 from E:\Tools\Solar\inverter-connect\venv\lib\site-packages\pip (python 3.10)
```

**Steps to reproduce:**
- Create venv
- Install dependencies using `pip install -r requirements.txt`

**Current behaviour:**
Installation fails with
```
ERROR: In --require-hashes mode, all requirements must have their versions pinned with ==. These do not:
    colorama from https://files.pythonhosted.org/packages/d1/d6/3965ed04c63042e047cb6a3e6ed1a63a35087b6a609aa3a15ed8ac56c221/colorama-0.4.6-py2.py3-none-any.whl (from click==8.1.7->-r .\requirements.txt (line 25))
```

**Expected behaviour:**
Installation of requirements completes successfully.

prodigy, 2024-02-11:
I just noticed I do not need to manually call pip install, nonetheless the installation fails with:
```
+ E:\Tools\Solar\inverter-connect\.venv-app\Scripts\inverter_app edit-settings

Traceback (most recent call last):
  File "C:\Users\sgrun\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\sgrun\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "E:\Tools\Solar\inverter-connect\.venv-app\Scripts\inverter_app.exe\__main__.py", line 4, in <module>
  File "E:\Tools\Solar\inverter-connect\inverter\__main__.py", line 7, in <module>
    from inverter.cli import cli_app
  File "E:\Tools\Solar\inverter-connect\inverter\cli\cli_app.py", line 16, in <module>
    from cli_base.toml_settings.api import TomlSettings
  File "E:\Tools\Solar\inverter-connect\.venv-app\lib\site-packages\cli_base\toml_settings\api.py", line 11, in <module>
    from cli_base.cli_tools.path_utils import backup, expand_user
  File "E:\Tools\Solar\inverter-connect\.venv-app\lib\site-packages\cli_base\cli_tools\path_utils.py", line 4, in <module>
    import pwd
ModuleNotFoundError: No module named 'pwd'
PS E:\Tools\Solar\inverter-connect>
```

jedie, 2024-02-13:
> Using a Ubuntu in WSL works.

That's it.

`pwd` doesn't exists under windows. I did not use windows. It would be too much work for me to customize everything for windows. But basically it could work.

closed 2024-02-13 by jedie