jedie/inverter-connect #134: Does not work since Firmware Update

Tixav, 2024-08-29:
SInce my inverter is update itsef to MW3_SSL_5408_1.0B (by OTA, I have SUN1000 so 2x MPPT) on the August 13th then nothing works at all, I read (on many geman post) that Modus port still works (8899) so I use solarman_mqtt and its works ! but I cannot use with home assistant or domoticz via MQTT. So I think finally the cause is  : the port 48899 is closed

CarstenGrohmann, 2024-08-30:
Both ports are used with different protocols:

-  8899 uses Modbus/TCP
- 48899 uses Modbus/AT

If you switch the port, you've also to switch the communication protocol also. I'm not sure, if inverter-connect supports port 8899 with Modbus over TCP.

What do you want? Reading your inverter data and publish them to HA?

Tixav, 2024-08-30:
I already try by changing 48899 by 8899 in the /root/.config/inverter-connect/inverter-connect.toml but nothing to do -> timeout
I check in the code "connection.py /  API.py...", I do not find a choice of protocol, I see  just AT protocol only (I have only few basic knowledge in python :-(  )
At the end I want to read data from inverter to domoticz via MQTT (homeassistant topic, by using MQTT auto discovery hardware)
until the August 13th, before the automatic upgrade of the inverter that was working well
Thanks for you help

CarstenGrohmann, 2024-08-31:
As you goal is to use domoticz with MQTT auto discovery you could try another project? 

I own a different inverter (Deye SUN300G3-EU-230) and I'm very satisfied with https://github.com/kbialek/deye-inverter-mqtt/ in combination with my self written plugin for Home Assistent auto discovery https://github.com/CarstenGrohmann/deye-mqtt-ha-plugin.

Perhaps that would also be an option for you?

Tixav, 2024-08-31:
I will try. I have deye sun1000g3-eu-230. I think it Can work. Thanks a lot. I am reporting son to you. Regards.

Tixav, 2024-09-13:
for your information I move to https://github.com/kbialek/deye-inverter-mqtt then I map MQTT to Domoticz with a plugin  MQTTMAPPER
Thanks your advice allows me to find a solution, thanks a lot